jneilliii/OctoPrint-TerminalCommandsExtended #1: Buttons Don't Add Right Away - Firefox

jneilliii, 2020-02-03:
From different repo by @EagleB3:

Hello! First, about the work of plugins.
Today I noticed the strange behavior of the "Terminal Commands Extended" plugin (using Mozilla Firefox 72.0.2 64bit).
There is no reaction to the "**+ Add**" button in the interface.
But if you press it (say, three times), then save the work results (by clicking the "**Save**" button) and refresh the page (**Ctrl+F5**), then three empty buttons will appear.
These buttons can be edited, and after that they will be displayed on the Terminal tab.
With the removal of empty (not edited buttons) the interface behavior is similar. Clicking on the red field (**x**) in the description of the button does not delete it; it will disappear only after the "**Save**" button is saved and the interface is updated by F5.
But with the edited buttons (which have contents in the Label and Command fields), this does not happen - after clicking on the red field (**x**) in the description of the button, such a button disappears immediately.

The interface of the **BedLevel Visualizer** plugin regarding adding / removing buttons **works correctly, without complaints**.

Yesterday I performed “Update-Upgrade-Clean” on my RPi3, I noticed the effect today. But, as they say in detective novels, "after that - does not mean that as a result of this." Perhaps the reason is this, but... I don’t know. Not sure.
[image]

jneilliii, 2020-02-03:
@EagleB3, does this happen only in Firefox? I do all my dev work in Chrome, so it might be a browser related issue I need to look into.

EagleB3, 2020-02-04:
Of course. One problem - I do not know exactly what is of interest for debugging. If you need to select some kind of event log, tell me more precisely which one and how to do it.
[image]
[image]

jneilliii, 2020-02-04:
Thanks, that was perfect.  I made a change just prior to pushing the version update that appears to have broke some stuff there.  I've just released a new version that will both fix bad data in the current config and resolve the issue with adding buttons moving forward.  Go into Software Update and force a check under the advanced options at the bottom. 

[image]

closed 2020-02-04 by jneilliii