jneilliii/OctoPrint-TerminalCommandsExtended #2: Easier icon select?

BerndJM, 2020-02-07:
First at all: a very nice plugin!

Just a suggestion (maybe for a next release?): 
It would be handy to have a search field on the icons select as it is e.g. in the 
TabOrder-Plugin
[image]
This would make the icon select much easier...

jneilliii, 2020-02-07:
Yeah, there were some technical glitches with the implementation in this plugin that I haven't been able to figure out yet. Will definitely keep trying.

jneilliii, 2020-02-08:
Just released version 0.1.3 that resolves this by moving the icon selector to the editor window in lieu of the drop-down arrow button option.

closed 2020-02-08 by jneilliii