jneilliii/OctoPrint-TerminalCommandsExtended #4: Send command while printing (Feature request)

pboucher14, 2020-02-10:
Would it be possible to activate a button while a print is in progress?

A setting could be added to specify if a button must be enabled while printing.
I'm asking because sometimes when a spool is almost empty, i  send the command M600
to start a filament change.  I created a button for that command but while i'm printing it is disabled.

jneilliii, 2020-02-10:
I had that in originally, but didn't think it worked so took it out. I'll add it back and verify it works before posting an updated version.

pboucher14, 2020-02-10:
Thank you, i will be watching for that.

Patrice Boucher

Envoyé le :10 février 2020 13:36
À : jneilliii/OctoPrint-TerminalCommandsExtended<mailto:OctoPrint-TerminalCommandsExtended@noreply.github.com>
Cc : Patrice Boucher<mailto:patrice_boucher@msn.com>; Author<mailto:author@noreply.github.com>

jneilliii, 2020-02-11:
Just pushed an update.  You may have to force check in Software Update in settings.

[image]

closed 2020-02-11 by jneilliii

pboucher14, 2020-02-13:
Thank you very much.
I tried it and it works perfectly