jneilliii/OctoPrint-TerminalCommandsExtended #5: After Uninstal with cleanup and reinstal the plugin does not works

pferlini63, 2020-03-19:
Dear Jneillii,
thanks for your work on this plugin. Its very great.

I uninstalled the previous version (updated) because one button with parameters does not works (give an error 500 ).

So i decided to reinstall from scratch the plugin.

This is the result when refreshing the page.

Note that there is an "empty" button and no save is possible nor modify 

Thanks a lot for your suggestion on how to proceed.

Paolo

packed_core.js?3514a079:17295 Could not bind view model terminalcommandsextendedViewModel to target #control-terminal-custom : TypeError: Unable to process binding "visible: function(){return loginStateViewModel.isUser && settingsViewModel.settings.plugins.terminalcommandsextended.commands().length > 0 }"
Message: Cannot read property 'commands' of undefined
    at visible (eval at parseBindingsString (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11374:90), <anonymous>:3:157)
    at update (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11409:229)
    at a.B.i (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11378:286)
    at Function.Pc (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11357:355)
    at Function.Qc (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11357:133)
    at Function.aa (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11356:482)
    at Object.a.m.a.B (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11355:39)
    at http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11378:273
    at Object.q (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11317:244)
    at m (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11377:318)
    at k (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11375:387)
    at Object.a.Rb (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11381:413)
    at http://192.168.50.150/static/webassets/packed_core.js?3514a079:17288:40
    at Pn (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11208:530)
    at Function.<anonymous> (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11229:66)
    at http://192.168.50.150/static/webassets/packed_core.js?3514a079:17258:31
    at Pn (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11208:530)
    at Function.<anonymous> (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11229:66)
    at bindViewModels (http://192.168.50.150/static/webassets/packed_core.js?3514a079:17225:19)
    at sentryWrapped (http://192.168.50.150/static/webassets/packed_core.js?3514a079:12054:39315)
(anonymous) @ packed_core.js?3514a079:17295
Pn @ packed_libs.js?3339d949:11208
(anonymous) @ packed_libs.js?3339d949:11229
(anonymous) @ packed_core.js?3514a079:17258
Pn @ packed_libs.js?3339d949:11208
(anonymous) @ packed_libs.js?3339d949:11229
bindViewModels @ packed_core.js?3514a079:17225
sentryWrapped @ packed_core.js?3514a079:12054
setTimeout (async)
(anonymous) @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:17377
fire @ packed_libs.js?3339d949:3269
fireWith @ packed_libs.js?3339d949:3399
done @ packed_libs.js?3339d949:9306
(anonymous) @ packed_libs.js?3339d949:9549
sentryWrapped @ packed_core.js?3514a079:12054
load (async)
T @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:12054
send @ packed_libs.js?3339d949:9601
ajax @ packed_libs.js?3339d949:9207
OctoPrintClient.ajax @ packed_client.js?313d9a0c:145
OctoPrintClient.get @ packed_client.js?313d9a0c:158
OctoPrintSettingsClient.get @ packed_client.js?313d9a0c:1547
SettingsViewModel.self.requestData @ packed_core.js?3514a079:5416
fetchSettings @ packed_core.js?3514a079:17354
sentryWrapped @ packed_core.js?3514a079:12054
setTimeout (async)
(anonymous) @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:17448
fire @ packed_libs.js?3339d949:3269
fireWith @ packed_libs.js?3339d949:3399
done @ packed_libs.js?3339d949:9306
(anonymous) @ packed_libs.js?3339d949:9549
sentryWrapped @ packed_core.js?3514a079:12054
load (async)
T @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:12054
send @ packed_libs.js?3339d949:9601
ajax @ packed_libs.js?3339d949:9207
OctoPrintClient.ajax @ packed_client.js?313d9a0c:145
OctoPrintClient.ajaxWithData @ packed_client.js?313d9a0c:154
OctoPrintClient.post @ packed_client.js?313d9a0c:166
OctoPrintClient.postJson @ packed_client.js?313d9a0c:179
OctoPrintBrowserClient.passiveLogin @ packed_client.js?313d9a0c:820
LoginStateViewModel.self.requestData @ packed_core.js?3514a079:3317
onServerConnect @ packed_core.js?3514a079:17420
(anonymous) @ packed_core.js?3514a079:17445
fire @ packed_libs.js?3339d949:3269
fireWith @ packed_libs.js?3339d949:3399
deferred.<computed> @ packed_libs.js?3339d949:3737
DataUpdater.self._onConnectMessage @ packed_core.js?3514a079:15014
(anonymous) @ packed_client.js?313d9a0c:421
Pn @ packed_libs.js?3339d949:11208
(anonymous) @ packed_libs.js?3339d949:11229
OctoPrintSocketClient.propagateMessage @ packed_client.js?313d9a0c:420
(anonymous) @ packed_client.js?313d9a0c:515
(anonymous) @ packed_libs.js?3339d949:11225
_t @ packed_libs.js?3339d949:11213
(anonymous) @ packed_libs.js?3339d949:11224
(anonymous) @ packed_libs.js?3339d949:11229
onMessage @ packed_client.js?313d9a0c:514
EventTarget.dispatchEvent @ packed_libs.js?3339d949:22600
(anonymous) @ packed_libs.js?3339d949:23313
SockJS._transportMessage @ packed_libs.js?3339d949:23311
EventEmitter.emit @ packed_libs.js?3339d949:22516
WebSocketTransport.ws.onmessage @ packed_libs.js?3339d949:25387
Show 33 more frames
packed_core.js?3514a079:17277 Did not bind view model terminalcommandsextendedViewModel to target #terminalcommandsextended since it does not exist

jneilliii, 2020-03-19:
Did you restart OctoPrint both when you uninstalled and then after you installed again?  The error it is showing doesn't make much sense because the default `commands` is in initialization. Maybe force a fresh of your browser by either pressing ctrl+f5 or shift+f5 in your browser.

pferlini63, 2020-03-20:
Dear Jneillii,
i redo for to be sure that all procedure was as you mentioned before.
1) cleanup & unistall plugin
2) reboot system
3) installation of plugin
4) reboot
5) reload page (shift+F5 & ctrl+F5)
6) test

the result is shown below (unfortunatly doesn't change).

Thanks in advance 

Paolo

P.S. im using chrome & Opera and the result is the same for both 

Could not bind view model terminalcommandsextendedViewModel to target #control-terminal-custom : TypeError: Unable to process binding "visible: function(){return loginStateViewModel.isUser && settingsViewModel.settings.plugins.terminalcommandsextended.commands().length > 0 }"
Message: Cannot read property 'commands' of undefined
    at visible (eval at parseBindingsString (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11374:90), <anonymous>:3:157)
    at update (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11409:229)
    at a.B.i (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11378:286)
    at Function.Pc (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11357:355)
    at Function.Qc (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11357:133)
    at Function.aa (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11356:482)
    at Object.a.m.a.B (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11355:39)
    at http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11378:273
    at Object.q (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11317:244)
    at m (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11377:318)
    at k (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11375:387)
    at Object.a.Rb (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11381:413)
    at http://192.168.50.150/static/webassets/packed_core.js?3514a079:17288:40
    at Pn (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11208:530)
    at Function.<anonymous> (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11229:66)
    at http://192.168.50.150/static/webassets/packed_core.js?3514a079:17258:31
    at Pn (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11208:530)
    at Function.<anonymous> (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11229:66)
    at bindViewModels (http://192.168.50.150/static/webassets/packed_core.js?3514a079:17225:19)
    at sentryWrapped (http://192.168.50.150/static/webassets/packed_core.js?3514a079:12054:39315)
(anonymous) @ packed_core.js?3514a079:17295
Pn @ packed_libs.js?3339d949:11208
(anonymous) @ packed_libs.js?3339d949:11229
(anonymous) @ packed_core.js?3514a079:17258
Pn @ packed_libs.js?3339d949:11208
(anonymous) @ packed_libs.js?3339d949:11229
bindViewModels @ packed_core.js?3514a079:17225
sentryWrapped @ packed_core.js?3514a079:12054
setTimeout (async)
(anonymous) @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:17377
fire @ packed_libs.js?3339d949:3269
fireWith @ packed_libs.js?3339d949:3399
done @ packed_libs.js?3339d949:9306
(anonymous) @ packed_libs.js?3339d949:9549
sentryWrapped @ packed_core.js?3514a079:12054
load (async)
T @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:12054
(anonymous) @ packed_core.js?3514a079:12054
send @ packed_libs.js?3339d949:9601
ajax @ packed_libs.js?3339d949:9207
OctoPrintClient.ajax @ packed_client.js?313d9a0c:145
OctoPrintClient.get @ packed_client.js?313d9a0c:158
OctoPrintSettingsClient.get @ packed_client.js?313d9a0c:1547
SettingsViewModel.self.requestData @ packed_core.js?3514a079:5416
fetchSettings @ packed_core.js?3514a079:17354
sentryWrapped @ packed_core.js?3514a079:12054
setTimeout (async)
(anonymous) @ packed_core.js?3514a079:12054
[... 40 lines cut ...]
Show 33 more frames
packed_core.js?3514a079:17277 Did not bind view model terminalcommandsextendedViewModel to target #terminalcommandsextended since it does not exist
packed_core.js?3514a079:17295 Could not bind view model terminalcommandsextendedViewModel to target #settings_plugin_terminalcommandsextended : TypeError: Unable to process binding "sortable: function(){return { data:settingsViewModel.settings.plugins.terminalcommandsextended.commands,options:{ cancel:'.unsortable',handle:'.move_command'}} }"
Message: Cannot read property 'commands' of undefined
    at sortable (eval at parseBindingsString (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11374:90), <anonymous>:3:135)
    at init (http://192.168.50.150/static/webassets/packed_plugins.js?fe12a96d:9069:40)
    at http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11378:1
    at Object.w (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11345:392)
    at http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11377:503
    at Object.q (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11317:244)
    at m (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11377:318)
    at k (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11375:387)
    at g (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11375:278)
    at k (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11375:440)
    at g (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11375:278)
    at k (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11375:440)
[... cut ...]
    at Function.<anonymous> (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11229:66)
    at http://192.168.50.150/static/webassets/packed_core.js?3514a079:17258:31
    at Pn (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11208:530)
    at Function.<anonymous> (http://192.168.50.150/static/webassets/packed_libs.js?3339d949:11229:66)
    at bindViewModels (http://192.168.50.150/static/webassets/packed_core.js?3514a079:17225:19)
    at sentryWrapped (http://192.168.50.150/static/webassets/packed_core.js?3514a079:12054:39315)

jneilliii, 2020-03-20:
This is very strange to me because I've done the same exact steps as mentioned above on two different OctoPrint installations and this doesn't happen for me. Are there any errors in octoprint.log on first starting after the reinstall/reboot process?

jneilliii, 2020-03-20:
You could try installing the older version, add a command and then upgrade and it might work. Use the URL below in plugin manager after uninstall/cleanup/restart. Add command, then upgrade.

https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/0.1.1.zip

pferlini63, 2020-03-20:
Hi,
done without to much success .... 

i've uninstalled the plugin, rebooted raspi and then followed you instructions with this results:

pi@raspberrypi:~ $ pip install "https://github.com/jneilliii/OctoPrint-TerminalC                ommandsExtended/archive/0.1.1.zip"
Collecting https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archi                ve/0.1.1.zip
  Downloading https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/ar                chive/0.1.1.zip
     \ 2.1MB 2.7MB/s
    Complete output from command python setup.py egg_info:
    Could not import OctoPrint's setuptools, are you sure you are running that u                nder the same python installation that OctoPrint is installed under?

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 255 in /tmp/pip-F4Ea32                -build/

jneilliii, 2020-03-20:
You didn't try to install using that url in plugin manager and instead via pip install in ssh? If you want to go that route you'll need to run the pip install command from within the octoprint virtual environment. I think it's as easy as doing the following, or even easier, just paste the URL into OctoPrint's Plugin Manager when you click Get More, enter the URL in the bottom and click install.

```
cd ~/oprint/bin
./pip install https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/0.1.1.zip
```

jneilliii, 2020-03-20:
Oh yeah, the 0.1.1 branch was broke. Try install using this one and let's see if it's the same issue or not. If it's still a problem I can look into this other error in octoprint.log.  That's the root cause of the problem I think.

https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/0.1.2.zip

jneilliii, 2020-03-20:
Oh wait, hold off.  I know what the problem is.

jneilliii, 2020-03-20:
No, never mind. I don't think it's what I thought. Try the install linked above and let's see if that works or not.

pferlini63, 2020-03-20:
... same result ....
I'm very sad, because i liked a lot your plugin. Any Idea ? Can i support you in a whatever way ?

Paolo

jneilliii, 2020-03-20:
Can you check your config.yaml file to see what's in there for this plugin?

jneilliii, 2020-03-20:
The very oldest/first version could be installed using the url below. Not sure it will help in this situation or not.

https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/070f404ce9ee33b6c5c03184be5c0d644b2c8a3a.zip

pferlini63, 2020-03-20:
Dear Jneilliii,
i reinstalled the version 0.1.2 as suggested (see below the installaztion log) and the config.yaml contains 

   firmwareupdater:
        avrdude_avrmcu: m2560
        avrdude_baudrate: ''
        avrdude_path: /usr/bin/avrdude
        avrdude_programmer: wiring
        dfuprog_avrmcu: ''
        enable_postflash_commandline: false
        enable_postflash_delay: true
        enable_postflash_gcode: true
        flash_method: avrdude
        postflash_commandline: ''
        postflash_delay: '2'
        postflash_gcode: M502;M500;
    simpleemergencystop: null
    softwareupdate:
        _config_version: 7
    **terminalcommandsextended: null**
    tracking:
        enabled: true
        unique_id: 1f9e9573-efec-4701-98f9-9ce07d45f59d
    yamlpatcher: null

```
Installing plugin...
Installing plugin from https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/0.1.2.zip...
/home/pi/OctoPrint/venv/bin/python2 -m pip --disable-pip-version-check install https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/0.1.2.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/0.1.2.zip
Downloading https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/0.1.2.zip
Requirement already satisfied: OctoPrint in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from Terminal-Commands-Extended==0.1.2) (1.4.0)
Requirement already satisfied: flask<0.13,>=0.12 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (0.12.5)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (2.8.1)
Requirement already satisfied: tornado==4.5.3 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (4.5.3)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (2018.8.29)
Requirement already satisfied: Flask-Login<0.5,>=0.4.1 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (0.4.1)
Requirement already satisfied: Flask-Babel<0.13,>=0.12 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (0.12.2)
Requirement already satisfied: Flask-Assets<0.13,>=0.12 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (0.12)
Requirement already satisfied: blinker<1.5,>=1.4 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (1.4)
Requirement already satisfied: werkzeug<0.17,>=0.16 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (0.16.1)
Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (0.1)
Requirement already satisfied: PyYAML<6,>=5.1 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (5.1)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (3.1.1)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (3.4)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (0.7.19)
Requirement already satisfied: watchdog<0.10,>=0.9.0 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (0.9.0)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (0.1.5.post0)
Requirement already satisfied: netifaces<0.11,>=0.10.9 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (0.10.9)
Requirement already satisfied: pylru<1.3,>=1.2 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (1.2.0)
Requirement already satisfied: rsa<5,>=4.0 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (4.0)
Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (1.5.0.1)
Requirement already satisfied: requests<3,>=2.22.0 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (2.22.0)
Requirement already satisfied: semantic_version<2.9,>=2.8 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (2.8.4)
Requirement already satisfied: psutil<5.7,>=5.6.5 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (5.6.7)
Requirement already satisfied: Click<8,>=7 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Terminal-Commands-Extended==0.1.2) (7.0)
[... cut ...]
Requirement already satisfied: six in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from websocket-client<0.57,>=0.56->OctoPrint->Terminal-Commands-Extended==0.1.2) (1.11.0)
Requirement already satisfied: pytz>=0a in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from Babel>=2.3->Flask-Babel<0.13,>=0.12->OctoPrint->Terminal-Commands-Extended==0.1.2) (2018.5)
Installing collected packages: Terminal-Commands-Extended
Running setup.py install for Terminal-Commands-Extended: started
Running setup.py install for Terminal-Commands-Extended: finished with status 'done'
Successfully installed Terminal-Commands-Extended-0.1.2
Done!
```

pferlini63, 2020-03-20:
Dear Jneilliii, many thanks for you support and suggestions.

I edited config.yaml, removing the row `terminalcommandsextended: null` now all works like a charms.

Thanks a lot again.

Paolo

closed 2020-03-20 by pferlini63

jneilliii, 2020-03-21:
Good news, glad to help.