jneilliii/OctoPrint-TerminalCommandsExtended #7: [SUG] Parameters inside dialog/popup

randellhodges, 2020-04-03:
Loving the buttons. I'm also using them with your Bed Visualizer plugin. I'd love to see this rolled in system wide somehow.

A feature that would be nice is to be able to specify parameters in a popup/dialog when the button is pressed. Press button, dialog opens with some text and parameters (including checkboxes/dropdown lists would be awesome) and then press ok/cancel etc.

jneilliii, 2020-04-03:
So you can configure input parameters for the buttons now, you mean to extend this as an optional pop-up form? Anything is possible with enough development time (limited for me) and interest.

randellhodges, 2020-04-03:
Exactly. I currently have a couple of buttons for PID tuning since I never remember to store these in the firmware or make note of them. Or even loading/saving mesh. By having an option to hide the parameters inside a dialog, the less used buttons wouldn't need to clutter up the page space.

Instead of having a couple of buttons, having one, that when I click it pops up a dialog with a dropdown of "Extruder" and "Bed" and to then an input box where I could put the temperature would be the end goal.

An intermediate step of just being able to specify the parameters in a dialog would be great. Defining parameters as a dropdown or checkbox would/could be a separate feature.