jneilliii/OctoPrint-TerminalCommandsExtended #9: [Feature Request] Import/Export of buttons

KenLucke, 2020-06-01:
I would love to be able to export the buttons to a CSV file or similar, to be able to back them up independently of the main OP backup, and to be able to import them into another instance of OP running on another machine without having to individually enter all the same buttons all over again.  It would also enable creating/editing the buttons in the CSV file instead of the plugin interface.

EagleB3, 2020-09-01:
@jneilliii , Can you please tell me which file the plugin currently stores information about the created buttons in?
So that you can back up your settings.

jneilliii, 2020-09-01:
All configurations are stored in config.yaml and as such, the standard OctoPrint Backup Plugin will back those up.

EagleB3, 2020-09-01:
@jneilliii , This is very reasonable and very convenient.
Thank!

Jon-So, 2020-12-02:
following!

Jon-So, 2020-12-02:
anybody have a good list of commands and willing to share them?

KenLucke, 2020-12-02:
> anybody have a good list of commands and willing to share them?

[image]

```
    terminalcommandsextended:
        _config_version: 4
        commands:
        -   command: 'G28
            G1 X0 Y0 Z0'
            confirmation: true
            enabled_while_printing: false
            icon: fas fa-home
            input: []
            label: Home Print Head
            message: You are about to initated a Print Head Home to 0,0,0
            offset: '0'
            tooltip: Home Print Head to 0,0,0
            width: '6'
        -   command: 'G28
            G29 T S35000'
            confirmation: true
            enabled_while_printing: false
            icon: fas fa-bed
            input: []
            label: Bed Level
            message: You are about to initated a Bed Level Probe Sequence
            offset: '0'
            tooltip: Level Bed with BLToucch Probe
            width: '6'
        -   command: ''
            confirmation: false
            enabled_while_printing: false
            icon: fas fa-gear
            input: []
            label: <BR>
            message: ''
            offset: 0
            tooltip: ''
            width: 12
        -   command: ''
            confirmation: false
            enabled_while_printing: false
            icon: fas fa-gear
            input: []
            label: <BR>
            message: ''
            offset: 0
            tooltip: ''
            width: 12
        -   command: M290 Z0.10 P0
            confirmation: false
            enabled_while_printing: true
            icon: fas fa-angle-up
            input: []
            label: Babystep Z +0.10
            message: ''
            offset: '1'
            tooltip: Babystep Z +0.1
            width: '2'
        -   command: M290 Z0.25 P0
            confirmation: false
            enabled_while_printing: true
            icon: fas fa-angle-double-up
            input: []
            label: Babystep Z +0.25
            message: ''
            offset: '0'
            tooltip: Babystep Z +0.25
            width: '2'
[... 627 lines cut ...]
            offset: '0'
            tooltip: PID Tune Build Plate 80°/10 Iterations
            width: '2'
        -   command: M303 E-1 S90 C10
            confirmation: true
            enabled_while_printing: false
            icon: fab fa-hotjar
            input: []
            label: PID Tune Bed 90°/10
            message: Are you sure that you wish to PID Tune the Buildplate Now?
            offset: '0'
            tooltip: PID Tune Build Plate 90°/10 Iterations
            width: '2'
        -   command: M303 E-1 S100 C10
            confirmation: true
            enabled_while_printing: false
            icon: fab fa-hotjar
            input: []
            label: PID Tune Bed 100°/10
            message: Are you sure that you wish to PID Tune the Buildplate Now?
            offset: 0
            tooltip: PID Tune Build Plate 100°/10 Iterations
            width: '2'

```

jneilliii, 2020-12-02:
Good idea, I wonder if I could open up the wiki here for everyone to edit....

Jon-So, 2020-12-03:
@KenLucke Thank you!