jneilliii/OctoPrint-TerminalCommandsExtended #10: [FR] basic layout with <BR/>

Andrei-Pozolotin, 2020-06-04:
perhaps it would be easy to implement an extra "separator button" with `<br/>`
which would help to preserve button layout during screen rotation with TouchUI

jneilliii, 2020-06-04:
Andrei, I notice a good amount of FRs from you today, I'm glad you're submitting ideas, maybe you would consider becoming a Patron?

https://www.patreon.com/jneilliii?fan_landing=true

Andrei-Pozolotin, 2020-06-04:
ok, now I am

Andrei-Pozolotin, 2020-06-20:
1. so, I have about 40 custom commands now:
[image]

2. w/o layout control this great feature quickly deteriorates into unusable mess

3. :-)
[image]

jneilliii, 2020-06-20:
LOL, I'm still trying to figure out the best way to handle this.

EagleB3, 2020-08-31:
The <BR> function would be awesome!

So far I have applied this solution: I made long buttons. See screenshot:
[image]

KenLucke, 2020-09-06:
Just noticed this thread.  I spaced mine using Option-Spaces (on a Mac, not sure what the equivalent on a PC is [Alt-Space, maybe?]), or "Hard [Non-Breaking] Spaces" (one problem, cutting and pasting Option-Spaces into a new button translates them into normal spaces, which are ignored in quantities >1 in html)

jneilliii, 2020-09-06:
I just pushed an update to my next dev version that will now recognize the text `<BR>` in a command label and replace that with a html break when rendered. This is just a quick fix until I can implement a better approach, potentially using an additional add button for adding a break or divider instead. This would allow for also rendering it properly in the settings UI, so expect a future update with a new add button.

If you want to try it out you can install by copying/pasting the URL below into plugin manager > get more > ...from URL and clicking Install.

```
https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/0.1.6.zip
```

KenLucke, 2020-09-06:
> If you want to try it out you can install by copying/pasting the URL below into plugin manager > get more > ...from URL and clicking Install.

Cool.  Will do as soon as current print on one machine is over, later today.  Other machines have prints ranginng from four more days to 11 more days :)

In contemplating what you are doing with be buttons, how about creating a [user definable] table structure (i.e., 2 column, 4 column, etc., and spaced based on max width of largest button, plus small buffer) for the buttons with the button centered in each cell?  That way, they're all automatically spaced and structured.

jneilliii, 2020-09-06:
That would be a full re-write unfortunately. I just pushed the update with the add break button option mentioned above to the same branch. It's a little tricky with the settings because it is rendered a little differently to be able to be sortable, but you can move the break around, add additional breaks in a row to give more space, etc.

[image]

[image]

KenLucke, 2020-09-06:
> That would be a full re-write unfortunately.

That's too bad.  Maybe if you ever visualize doing that full rewrite :)

In the meantime, your current tweak will certainly make things easier.  Thanks!

jneilliii, 2020-09-06:
Yeah, I wish I had more time to work on plugins, but full time job and such, hours are limited. Was just hoping to get you guys something that I could quickly put together in the meantime as a temporary stop gap.

jneilliii, 2020-09-06:
@KenLucke thanks for the sponsorship, your support is greatly appreciated and your level will definitely influence this feature request. If you'd like to be listed on my plugin's repositories please let me know, as that is also a benefit to your tier. I can link it to your own personal web page or github user profile, or just list your name, it's up to you. 

Since the OctoPrint's layout is already supporting bootstrap 2 scaffolding I don't think it would be too hard to add a width and offset options to the command editor for each that incorporates the correct span1, span2, ...span12 and offset1, offset2... classes and give you that kind of control of width and positioning. More details about bootstrap 2 scaffolding can be found [here](https://getbootstrap.com/2.3.2/scaffolding.html#gridSystem). Is that what you were kind of thinking?

KenLucke, 2020-09-06:
> Yeah, I wish I had more time to work on plugins, but full time job and such, hours are limited. Was just hoping to get you guys something that I could quickly put together in the meantime as a temporary stop gap.

I understand how life affects hobbies :).  No worries, I appreciate the work you DO put in.

KenLucke, 2020-09-06:
> @KenLucke thanks for the sponsorship, your support is greatly appreciated and your level will definitely influence this feature request. If you'd like to be listed on my plugin's repositories please [...]

I didn't do it for recognition, which is why I selected "private" :)

KenLucke, 2020-09-06:
> Since the OctoPrint's layout is already supporting bootstrap 2 scaffolding I don't think it would be too hard to add a width and offset options to the command editor for each that incorporates the [...]

That looks ideal.

jneilliii, 2020-09-06:
> which is why I selected "private" :) 

Sorry I missed that. 

> before HTML 4, even

Me too to be honest, I just know about this one specific from so many years of octoprint plugin development. It also mimics the undocumented capabilities of standard custom controls, as that also has width and offset.

KenLucke, 2020-09-07:
Uh, yeah,  OK.  :)

jneilliii, 2020-09-07:
lol. yeah, just notes for me really @KenLucke. I think I've made some progress, just have a couple of more things to figure out, specifically being able to re-order items. There's a good chance all the work I've done last night and today will have to be entirely redone to make that work I fear. If that is the case, I may re-think the whole thing and probably make a full panel/group system where you specify the panel width/offset and add buttons to each of them, but I may just be over thinking it.

This is what I got to though and these updates are now on that same 0.1.6 dev branch linked above.

[image]

[image]

KenLucke, 2020-09-07:
Glad you were just spitballing to yourself, cause I was way out in the wilderness and had no way to get back on all that `<grin>`

Andrei-Pozolotin, 2020-12-11:
I tried 0.1.6 and:
* I can click "Add Break"
* break button does show up
* drag and drop does no work for the "break button"
* drag and drop does no work for all existing buttons
do I have do re-create existing buttons for the new version?

jneilliii, 2020-12-11:
There are still some weird things with this version relative to dragging and layout that I haven't been able to work on yet. This may be related.

closed 2022-05-28 by jneilliii