jneilliii/OctoPrint-TerminalCommandsExtended #12: [FR] button to hide the "hide"

Andrei-Pozolotin, 2020-06-04:
it would help to introduce an extra button next to "Auto Scroll" 
which would hide the default panel with:
* "Hide temperature messages"
* "Advanced options"
* etc

jneilliii, 2020-06-04:
Maybe move those under the advanced section?

Andrei-Pozolotin, 2020-06-04:
"Auto Scroll" button already eats up whole one line space, 
my intent was to re-use that space if possible

jneilliii, 2020-06-04:
Sorry for the questions, just want to make sure I understand. So ability to add additional buttons within that space next to Auto Scroll or just collapse everything so it's only the one line of space being taken up and therefore pushing the custom buttons up higher in the UI like this?

[image]

Andrei-Pozolotin, 2020-06-05:
I am not sure; how about this: 

1. say, there are 2 new sticky buttons: `Control` and `Filters`

2. `Control` enables panel with custom commands

3. `Filters` enables panel with existing checkboxes, etc

4. new sticky buttons remember their state after page reload

5. here is a layout sketch:
[ocotoprint-plugin.pdf](https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/files/4733553/ocotoprint-plugin.pdf)

jneilliii, 2020-06-05:
Gotcha. thanks so much for the sketch, that makes it much more clear.