jneilliii/OctoPrint-TerminalCommandsExtended #13: [FR] numeric keypad markup for command parameters

Andrei-Pozolotin, 2020-06-20:
1. command parameters are a great feature

[image]

2. however, on android, in TouchUI, when editing parameters, say `SIZE_X`
[image]

chrome opens "regular keyboard" 
[image]

instead of "numeric keypad"
[image]

3. at the same time, on the default "control page", for editing "Feed Rate"
chrome opens "numeric keypad"

4. so apparently some `css` markup is required 
to make command parameters use "numeric keypad" by default

jneilliii, 2020-06-20:
The problem with this is that not all parameters are numeric, so it is acting as desired. Users may want to send an M117 message for example, which is text based.

Andrei-Pozolotin, 2020-06-20:
ok, can we then have a check box with meaning: "treat this parameter as numeric"?

jneilliii, 2020-06-20:
I might be able to simplify it a bit and just detect if the default value is numeric and apply the special class that way.

Andrei-Pozolotin, 2020-06-20:
sounds great, thank you

jneilliii, 2020-06-20:
feedrate is a slider in the default interface, what is it converted to in TouchUI?

jneilliii, 2020-06-20:
nevermind, I figure it out. Looks like it's converted to a "number" input type.

jneilliii, 2020-06-20:
Releasing new version with this change.

closed 2020-06-20 by jneilliii

Andrei-Pozolotin, 2020-06-21:
0.1.5 works as expected on android. thank you