jneilliii/OctoPrint-TerminalCommandsExtended #15: [FR] auto magical command parameters

Andrei-Pozolotin, 2020-06-22:
1. is it feasible to expose to buttons auto magical command parameters?

2. use case: there is an attachment which in response to a button command
auto paints hotbed with a glue stick, but glue application area position and size
need to be provided manually, despite model gcode is loaded in octoprint
and, first layer bounding box information is apparently available via some api

3. here are some plugins which can access loaded model gcode data:
* built-in gcode viewer
* https://github.com/marian42/octoprint-preheat

4. parameters to extract from gcode:
* hotbed temp
* hotend temp
* chamber temp
* first layer bounding box
* total model height

jneilliii, 2020-06-22:
Not sure if this is feasible for a generic plugin like this one, assuming I'm understanding your correctly. The plugins that you are referring to are actually reading the gcode file directly and reacting to those, which again is not feasible for this plugin because this isn't a gcode file plugin but a generic gcode commands plugin.

Andrei-Pozolotin, 2020-06-22:
1. in this case, can you expose an api to subscribe to externally provided parameters?

2. that is so I can write a plugin which will publish these parameters for your plugin

jneilliii, 2020-06-22:
I still don't understand the use case here. Can you describe your end goal please?

Andrei-Pozolotin, 2020-06-22:
1. the use case is to extract command parameters from the g-code
  in addition to having to provide them manually with user UI

2. utility: this can be used to make commands model-aware, for example:

* **example 1**: replicate features of https://github.com/marian42/octoprint-preheat
  except feed intercepted temperature as command parameter, i.e.
  instead of multiple command buttons: "Preheat ABS", "Preheat PLA", ...
  the would be only one "Preheat Model" button, 
  which would translate to `klipper` command
  "SET_PREHEAT TOOL=HOTEND_TEMP=TTT", 
  where temperature value TTT would be extracted from M104 g-code of the model
  the problem with octoprint-preheat is that it translates M104 -> M104
  where instead there is a need to map M104 -> SET_PREHEAT, which is doing extra steps

* **example 2**: instead of having to provide 4 parameters manually
to the `klipper` glue application command (and read these values manually from slicer every time)
```
GLUON_DRAW INIT_X=NNN INIT_Y=NNN SIZE_X=NNN SIZE_Y=NNN
```
the origin and size would be extracted from gcode first layer bounding box

3. how about a PR? 
will you accept a PR which would basically be a copy-paste 
of g-code context extract as it is done in some other plugins, 
and said context will then be injected into this plugin
as automatically discovered command parameters?

say, auto magic parameters will appear with prefix `GCODE`
* GCODE_BOX_INIT_X ; bounding box origin
* GCODE_BOX_INIT_Y ; bounding box origin
* GCODE_BOX_SIZE_X ; bounding box size
* GCODE_BOX_SIZE_Y ; bounding box size
* GCODE_BOX_SIZE_Z ; bounding box size
* GCODE_TEMP_M104 ; https://marlinfw.org/docs/gcode/M104.html
* GCODE_TEMP_M109 ; https://marlinfw.org/docs/gcode/M109.html
* GCODE_TEMP_M140 ; https://marlinfw.org/docs/gcode/M140.html
* GCODE_TEMP_M190 ; https://marlinfw.org/docs/gcode/M190.html
* GCODE_TEMP_M141 ; https://marlinfw.org/docs/gcode/M141.html
* GCODE_TEMP_M191 ; https://marlinfw.org/docs/gcode/M191.html

4. temperature extract seems rather straightforward:
https://github.com/marian42/octoprint-preheat/blob/master/octoprint_preheat/__init__.py#L84

5. bounding box is already exposed as "dimensions" after "file load / model view" action in UI:
https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/util/gcodeInterpreter.py#L506

jneilliii, 2020-06-22:
I would accept a PR for this, always open for that.

Andrei-Pozolotin, 2020-06-22:
ok, great. please re-open.

Andrei-Pozolotin, 2020-06-23:
@jneilliii Jim: 
do you have a recommended setup for plugin development / testing?
any good tutorials / walkthroughs? please share links if any? thank you.

jneilliii, 2020-06-23:
The plugin tutorial [here](https://docs.octoprint.org/en/master/plugins/gettingstarted.html) is a good start. There's also a YouTube video in the code and chat series where Gina writes a plugin.