jneilliii/OctoPrint-TerminalCommandsExtended #16: Multiple commands separated by semicolon (sometimes) fails

ruedli, 2020-07-19:
I added an 

M92 E415; M500 

to set my E-Steps. This does not process the M500, nor does it process any "second" command, also tried e.g. M92 E415;M117 Set eSteps.

Strangely enough it DOES work for my "Load filament" command:
M701 E1; M400
This loads extruder 2 and then waits, or:

M701 E0; M400; M701 E1; M400; M701 E2; M400; M701 E3; M400; M701 E4; M400
to load all filaments of the MMU2.

jneilliii, 2020-07-19:
Semicolons are comments, but I question why you don't put them on separate lines? If it's critical maybe try with colon and leave out spaces?

ruedli, 2020-07-20:
Thnx @jneilliii for looking into this so quickly. You are absolutely right: multiple lines is the only way this can and should work. I was confused by another plugin ("Terminal Commands") - so without "extended", which worked with the semicolons. I "copied" the keyboard definitions from that plugin into the "extended" plugin.

I tired colons, they do not work, but this is not required, the one command per line works as specified.

I close my "issue", sorry for my confusion. Your plugin "terminal commands extended" is superior, with the parameter definition, I like this a lot!

closed 2020-07-20 by ruedli