jneilliii/OctoPrint-TerminalCommandsExtended #19: [Enhancement] Live update Settings Dialog?

KenLucke, 2020-11-23:
Love Love Love the new spacing options (!), but it's a pain to go back and forth between the settings window and the actual window to see results (Open settings > Open TCE Settings > Change spacings > Save Settings > Close Main Settings Window > View Terminal Window for Changes > Lather > Rinse > Repeat)

Sometime when you get around to it, could the Settings window be made to update live as the spacings are entered there to get a feel for  what the final version version will be?

KenLucke, 2020-11-23:
Nevmind :)

It's doing it now, it didn't when I first installed it.  Not sure why.

jneilliii, 2020-11-23:
lol, I was just thinking to myself that I thought it did that already.

closed 2020-11-23 by jneilliii

KenLucke, 2020-11-23:
> lol, I was just thinking to myself that I thought it did that already.

Yeah, don't know why it didn't work at first.

h=However, there IS a problem:  I no longer seem to be able to drag buttons around in the settings dialog.