jneilliii/OctoPrint-TerminalCommandsExtended #20: Unable to drag items in Settings

KenLucke, 2020-11-23:
Since the update, I cannot drag the buttons around in the settings dialog.  I can create them, delete them, and get tooltips and cursor changes when mousing over them, but clicking and dragging them has no effect.

Mac OSX 15.5.7, Firefox and Safari, OctoPrint 1.4.2, TCE v0.1.6, Python v2.7.16 (no, I haven't updated, as 3.x broke a bunch of favorite plugins and I downgraded)

jneilliii, 2020-11-24:
Are there any errors in the developer console on initial page load?

KenLucke, 2020-11-24:
Not sure if this is what you are  looking for, but clearing the console, then reloading OctoPrint and going to the plugins settings and TCE  dialog, this is the result  (Errors and Warnings only):

```
Content Security Policy: Ignoring ‘x-frame-options’ because of ‘frame-ancestors’ directive.
Content Security Policy: The page’s settings blocked the loading of a resource at inline (“script-src”). onloadwff.js:71:799505
Content Security Policy: The page’s settings blocked the loading of a resource at inline (“script-src”). 3 utils.js:35:9
Content Security Policy: The page’s settings blocked the loading of a resource at data: (“media-src”).
Some cookies are misusing the recommended “SameSite“ attribute 5
Cookie “session_P443” will be soon rejected because it has the “SameSite” attribute set to “None” or an invalid value, without the “secure” attribute. To know more about the “SameSite“ attribute, read https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite languages
Cookie “session_P443” will be soon rejected because it has the “SameSite” attribute set to “None” or an invalid value, without the “secure” attribute. To know more about the “SameSite“ attribute, read https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite languages
Cookie “session_P443” will be soon rejected because it has the “SameSite” attribute set to “None” or an invalid value, without the “secure” attribute. To know more about the “SameSite“ attribute, read https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite 2 printerprofiles
Cookie “session_P443” will be soon rejected because it has the “SameSite” attribute set to “None” or an invalid value, without the “secure” attribute. To know more about the “SameSite“ attribute, read https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite 2 appkeys
Cookie “session_P443” will be soon rejected because it has the “SameSite” attribute set to “None” or an invalid value, without the “secure” attribute. To know more about the “SameSite“ attribute, read https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite 2 backup
```

jneilliii, 2020-11-24:
Thanks, confirmed bug. Something must have got messed up with all the moving of components around. I will investigate further.

KenLucke, 2020-11-24:
Cool, looking forward to an update.  Easy to break stuff when adding new functionality.

jneilliii, 2020-11-24:
Found it, I removed the sortable binding, I think I just need to add it back [her](https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/blob/1570b31cd5aa7ef1d9dcfc4dd7cd032c664d22c9/octoprint_terminalcommandsextended/templates/terminalcommandsextended_settings.jinja2#L9) instead of using the foreach binding. I'll test this in a little bit and get back with you.

KenLucke, 2020-11-24:
> Found it, I removed the sortable binding, I think I just need to add it back [here](https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/blob/1570b31cd5aa7ef1d9dcfc4dd7cd032c664d22c9/octoprint_terminalcommandsextended/templates/terminalcommandsextended_settings.jinja2#L9) instead of using the foreach binding. I'll test this in a little bit and get back with you.

That was fast. Must have taken your speedy pills this morning. :)

jneilliii, 2020-11-24:
Yeah, this is going to be more involved than I originally thought. Because of the settings interface providing a preview of the ordering it's breaking the link to the underlying config settings. Not really sure the best way to move forward on this one, but need to check into the available options for the sorting. For now I think switching this back to a simple list rather than the preview is going to be the quick fix for ordering, but then has the negative side effect of not being able to see how your offsets/widths will look.

KenLucke, 2020-11-24:
I can live with that for now.  :)

jneilliii, 2020-11-24:
You can copy/paste the URL below in Plugin Manager > Get More > ...from URL and click Install to verify for more it's working again.

```
https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/0.1.7rc1.zip
```

KenLucke, 2020-11-24:
Will do, but all printers are engaged at this time, so it will be about 12  hours before one of  them is done to be able to do so.,

KenLucke, 2020-11-24:
I might fire up the spare Pi and update everything, then try it... but I'm working on Christmas decorations today... I have [literally] about 20,000 lights to put up.

KenLucke, 2020-11-25:
Sorry, no joy.  Same unable to drag buttons.

jneilliii, 2020-11-25:
My bad, Ken, I based that release on the wrong branch. It should be fixed now, so if you reinstall using the same URL above it should be fixed for real now.

KenLucke, 2020-11-26:
Dragging has been reenabled, but as you drag, the point where the button shows and the point where the cursor shows are not the same, making it difficult to determine where you are about to drop the button.  Takes some getting use to.

jneilliii, 2020-11-26:
Yeah, I noticed that on some buttons, but not all of them for some reason. Definitely needs some more debugging.

KenLucke, 2020-11-26:
Well, at least they are draggable, and I was able to get my layout done.  Now when the buttons match the width, it will look awesome.

jneilliii, 2020-11-26:
Dragging might be better in this version.

```
https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/0.1.7rc2.zip
```

KenLucke, 2020-11-26:
Dragging IS much better, and column sizing is beautiful!

[image]

KenLucke, 2020-11-26:
Only ONE last request on this one for now :)

When clicking  "Duplicate Button," could you duplicate it at that spot (immediately after the button being clicked to duplicate) instead of at the bottom ("Create Button/Break" would still make a new one at the bottom)?  It would eliminate much dragging and scrolling for those of us with large, complicated lists of buttons.

jneilliii, 2020-11-26:
That might be possible to achieve, just have to dig into the js code to figure out how to add the duplicate in at a specifici position of the observableArray. Might be tricky, but I'll take a look.

EagleB3, 2020-12-01:
FireFox v.83.0 x64, plugin 0.1.7rc2.zip

The "crosses" for dragging are hidden by objects of other buttons. Clearing the cache in the browser doesn't help.
[image]

jneilliii, 2020-12-01:
You don't have any widths defined do you @EagleB3?

EagleB3, 2020-12-01:
@jneilliii , I may not understand the question correctly, but it looks like the answer = "Yes". 
The labels overlap each other. In the screenshot, the cursor is hovering over the label "G30Z0", but its service fields remain closed by the label located to the right of the "G30Z0":
[image]

jneilliii, 2020-12-01:
Yeah the changes that I did to accommodate #21  has inadvertently made this happen I think.  I'll look into it further to try and get a happy medium. I think if your buttons width was defined it would spread those out more or this is a problem with firefox rendering.

[image]

EagleB3, 2020-12-01:
Yes it is. Here's how the buttons look in the settings for the different widths they define:
[image]

KenLucke, 2020-12-01:
The key is to set a minimum of 2-3 for width, drag them to where you want them, then set the width lower if need be (why you'd ever want a width of 1, I don't know, unless there was no text).  You could also temporarily increase the offset of the _next_ button to space it out, then reduce later.

Unless you set a  min width in the settings window, which would make it not represent the real window properly, I'm not sure that there is a solution for this.  One thing that might help is when a new button is created, pre-set a width of 2 or 3.

KenLucke, 2020-12-01:
The key is to set a minimum of 2-3 for width, drag them to where you want them, then set the width lower if need be (why you'd ever want a width of 1, I don't know, unless there was no text).  You could also temporarily increase the offset of the _next_ button to space it out, then reduce later.

@jneilliii, Unless you set a  min width in the settings window, which would make it not represent the real window properly, I'm not sure that there is a solution for this.  One thing that might help is when a new button is created, pre-set a width of 2 or an offset of 1.

jneilliii, 2020-12-01:
I do default the width to three when you add the button. I'm sure I'll be able to figure something out. The issue is all in the css and inline blocking versus not blocking. The way it is now (non-blocking) allows for the smaller widths to overlay each other. It will be tricky, but I bet I can figure some kind of css hack out for this.

jneilliii, 2020-12-01:
I do like the idea of handling width 0 as "just make it work like before" in #24

EagleB3, 2020-12-02:
Forgive my poor knowledge of English - I am not sure that I understood the essence of the discussed issue correctly. But I want to draw your attention to the fact that the appearance and position of the buttons within some limits can be seen immediately after creation / modification, even without saving the changes made to the settings.
Unfortunately, when invoking the settings window, the Octoprint interface always moves the window to the left and up, and does not allow you to move the open settings window around the screen. Still, you can see the final appearance of the buttons (especially if the buttons are compact!)
[image]
[image]

EagleB3, 2020-12-02:
Do you think it makes sense to ask Gina Häußge to make the settings window at least moveable on the screen? This feature would greatly simplify the issue of viewing the final location of the buttons, and perhaps completely solve it.
I am not sure that I can correctly and unambiguously describe this task... :(
[image]

jneilliii, 2020-12-02:
I think I understand what you mean. There is a way to do this programmaticaly, the UI Customizer plugin does it, I just need the time to work on this, I do have a full time job and do this in my spare time.

garurumoon, 2021-04-22:
Hello. I am experiencing the same issue as indicated in the original post. Is this a known issue? if not - how can I be helpful?

jneilliii, 2021-04-22:
@garurumoon could you test the latest rc version by changing the release channel for the plugin in OctoPrint's software update settings to `Release Candidate` and upgrade when prompted? I haven't had much time to work on this plugin, but think dragging may be working better in it, unless you have a plugin conflict that is loading a different jquery sortable library or not including it at all. If you are still experiencing issues with 0.1.7 rc2 I'll need a list of plugins that you have installed.

garurumoon, 2021-04-22:
The Release candidate worked, thank you.

LouDnl, 2021-08-04:
Sorry for replying to an old issue, but I still cannot move any buttons.
Plugin version is 0.1.6
Octopi version 0.17
Octoprint version 1.6.1

Other plugins:
<html>
<body>

OctoPrint: 1.6.1 Release Channel: StableMaintenance RCsDevel RCs Installed: 1.6.1Available: 1.6.1Release Notes: https://github.com/OctoPrint/OctoPrint/releases/tag/1.6.1
--
  | Arc Welder: Anti-Stutter: unknown Release Channel: StableMaintenance RCsDevel RCs Installed: ?Available: ?Unknown error while checking for update, please check the logs
  | Bed Visualizer: 1.1.0 Release Channel: StableRelease Candidate Installed: 1.1.0Available: 1.1.0 - Numpy RemovalRelease Notes: https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/releases/tag/1.1.0
  | BLTouch Plugin: 0.3.4 Release Channel: StableRelease Candidate Installed: 0.3.4Available: 0.3.4Release Notes: https://github.com/jneilliii/OctoPrint-BLTouch/releases/tag/0.3.4
  | Custom Control Editor Plugin: 0.2.2 Installed: 0.2.2Available: Version 0.2.2Release Notes: https://github.com/Salandora/octoprint-customControl/releases/tag/0.2.2
  | Dashboard Plugin: 1.18.3 Release Channel: StableRelease Candidate Installed: 1.18.3Available: 1.18.3Release Notes: https://github.com/j7126/OctoPrint-Dashboard/releases/tag/1.18.3
  | DisplayLayerProgress Plugin: 1.26.0 Installed: 1.26.0Available: V1.26.0Release Notes: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.26.0
  | Dragon Order: 0.1.7 Installed: 0.1.7Available: 0.1.7Release Notes: https://github.com/jneilliii/OctoPrint-DragonOrder/releases/tag/0.1.7
  | Filament Manager: 1.8.1 Installed: 1.8.1Available: V1.8.1Release Notes: https://github.com/OllisGit/OctoPrint-FilamentManager/releases/tag/1.8.1
  | File Check Plugin: 2021.2.23 Release Channel: StablePrerelease Installed: 2021.2.23Available: 2021.2.23Release Notes: https://github.com/OctoPrint/OctoPrint-FileCheck/releases/tag/2021.2.23
  | Firmware Check Plugin: 2021.5.31 Release Channel: StablePrerelease Installed: 2021.5.31Available: 2021.5.31Release Notes: https://github.com/OctoPrint/OctoPrint-FirmwareCheck/releases/tag/2021.5.31
  | Floating Navbar: 0.3.7 Release Channel: StableRelease Candidate Installed: 0.3.7Available: 0.3.7Release Notes: https://github.com/jneilliii/OctoPrint-FloatingNavbar/releases/tag/0.3.7
  | GCode System Commands: 1.0.1 Installed: 1.0.1Available: 1.0.1Release Notes: https://github.com/kantlivelong/OctoPrint-GCodeSystemCommands/releases/tag/1.0.1
  | Gcodebar Plugin: 0.1.5 Installed: 0.1.5Available: Python3 compabilityRelease Notes: https://github.com/mtowara/OctoPrint-Gcodebar/releases/tag/0.1.5
  | GcodeEditor Plugin: 0.2.12 Installed: 0.2.12Available: 0.2.12Release Notes: https://github.com/ieatacid/OctoPrint-GcodeEditor/releases/tag/0.2.12
  | M73 Progress Plugin: 0.2.1 Installed: 0.2.1Available: 0.2.1Release Notes: https://github.com/cesarvandevelde/OctoPrint-M73Progress/releases/tag/v0.2.1
  | MeatPack - Automatic G-Code Compression: 1.5.21 Installed: 1.5.21Available: Version 1.5.21 - Marlin FixesRelease Notes: https://github.com/scottmudge/OctoPrint-MeatPack/releases/tag/v1.5.21
  | MultiCam: 0.2.8 Installed: 0.2.8Available: Update Version InformationRelease Notes: https://github.com/mikedmor/OctoPrint_MultiCam/releases/tag/0.2.8
  | Navbar Temperature Plugin: 0.14 Installed: 0.14Available: 0.14 Milestone releaseRelease Notes: https://github.com/imrahil/OctoPrint-NavbarTemp/releases/tag/0.14
  | Octolapse: 0.4.1 Release Channel: StableMaintenance RCsDevel RCs Installed: 0.4.1Available: v0.4.1Release Notes: https://github.com/FormerLurker/Octolapse/releases/tag/v0.4.1
  | Pi Support Plugin: 2021.8.2 Release Channel: StablePrerelease Installed: 2021.8.2Available: 2021.8.2Release Notes: https://github.com/OctoPrint/OctoPrint-PiSupport/releases/tag/2021.8.2
  | PortLister: 0.1.10 Installed: 0.1.10Available: 0.1.10Release Notes: https://github.com/markwal/OctoPrint-PortLister/releases/tag/0.1.10
  | Preheat Button: 0.8.0 Installed: 0.8.0Available: Add preheat on file select feature and fix a UI bug with UI CustomizerRelease Notes: https://github.com/marian42/octoprint-preheat/releases/tag/0.8.0
  | Print Time Genius Plugin: 2.2.8 Installed: 2.2.8Available: Fix missing print history and height calculationRelease Notes: https://github.com/eyal0/OctoPrint-PrintTimeGenius/releases/tag/2.2.8
  | PSU Control: 1.0.6 Installed: 1.0.6Available: 1.0.6Release Notes: https://github.com/kantlivelong/OctoPrint-PSUControl/releases/tag/1.0.6
  | PSU Control - RPi.GPIO: 1.0.4 Installed: 1.0.4Available: 1.0.4Release Notes: https://github.com/kantlivelong/OctoPrint-PSUControl-RPiGPIO/releases/tag/1.0.4
  | Resource Monitor: 0.3.4 Release Channel: StablePrerelease Installed: 0.3.4Available: 0.3.4Release Notes: https://github.com/Renaud11232/OctoPrint-Resource-Monitor/releases/tag/0.3.4
  | SnapStream: 0.3 Installed: 0.3Available: 0.3Release Notes: https://github.com/markwal/OctoPrint-SnapStream/releases/tag/0.3
  | Tab Order: 0.5.12 Release Channel: StableRelease Candidate Installed: 0.5.12Available: 0.5.12Release Notes: https://github.com/jneilliii/OctoPrint-TabOrder/releases/tag/0.5.12
  | Terminal Commands Extended: 0.1.6 Release Channel: StableRelease Candidate Installed: 0.1.6Available: 0.1.6Release Notes: https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/releases/tag/0.1.6
  | Themeify: 1.2.2 Installed: 1.2.2Available: v1.2.2Release Notes: https://github.com/Birkbjo/OctoPrint-Themeify/releases/tag/v1.2.2
  | Z Probe Offset Control: 0.5 Installed: 0.5Available: 0.5

</body>
</html>

jneilliii, 2021-08-04:
I think the issue for you might be the custom control editor plugin. We both use the same jquery draggable UI and the one in that repo is outdated and doesn't include everything. If it overwrites my import than it makes my plugins not work. You can disable that plugin once you have the settings configured the way you like it and the custom controls will remain, as that is part of core OctoPrint and the plugin just exposes the settings in the UI. It's either that are you are not running the release candidate version of the plugin that fixed it. Swith the release channel in the software update settings of OctoPrint for the plugin and set it to release candidate and upgrade when prompted.

LouDnl, 2021-08-05:
> I think the issue for you might be the custom control editor plugin. We both use the same jquery draggable UI and the one in that repo is outdated and doesn't include everything. If it overwrites my [...]

Thanks, will test that

robfrawley, 2021-08-28:
@jneilliii Anything in particular that you need help with to get such erroneous CSS definitions corrected (I don't want to take a look and put in time duplicating efforts if you are already actively working on such changes via a local dev branch)? If you'd like assistance expediting the RC version updates, feel free to let me know if any other components require updating (aside from the CSS) to correct this behavior and I'll likely have some extra time to take a look at it if needed.

Thanks again! All the best.

jneilliii, 2021-08-28:
@robfrawley if you want to take a look, by all means do. I'm not currently working on this plugin actively at the moment and if it's something that someone else could contribute that would be a huge help. The rc release of the plugin should match the rc branch here on the repo, so if you do decide to help make sure you're basing your work off that one. I'm also always open to improvements in code if you see anything done blatantly wrong or less efficient.

closed 2022-05-28 by jneilliii