jneilliii/OctoPrint-TerminalCommandsExtended #21: Button Widths not matching width settings

KenLucke, 2020-11-23:
Button widths are still depending on the text in the button, rather than the spacing set via the Width parameter in the button definition.

[image]

They should be taking up the number of columns that they are defined as, with the text/icon centered in the button.

Mac OSX 15.5.7, Firefox and Safari, OctoPrint 1.4.2, TCE v0.1.6, Python v2.7.16 (no, I haven't updated, as 3.x broke a bunch of favorite plugins and I downgraded)

KenLucke, 2020-11-23:
```
    terminalcommandsextended:
        _config_version: 4
        commands:
        -   command: 'G28

                G1 X0 Y0 Z0'
            confirmation: true
            enabled_while_printing: false
            icon: fas fa-home
            input: []
            label: Home Print Head
            message: You are about to initated a Print Head Home to 0,0,0
            offset: '0'
            tooltip: Home Print Head to 0,0,0
            width: '6'
        -   command: 'G28

                G29 T S35000'
            confirmation: true
            enabled_while_printing: false
            icon: fas fa-bed
            input: []
            label: Bed Level
            message: You are about to initated a Bed Level Probe Sequence
            offset: '0'
            tooltip: Level Bed with BLToucch Probe
            width: '6'
        -   command: M290 Z0.10 P0
            confirmation: false
            enabled_while_printing: true
            icon: fas fa-angle-up
            input: []
            label: Babystep Z +0.10
            message: ''
            offset: '0'
            tooltip: Babystep Z +0.1
            width: '4'
        -   command: M290 Z0.25 P0
            confirmation: false
            enabled_while_printing: true
            icon: fas fa-angle-double-up
            input: []
            label: Babystep Z +0.25
            message: ''
            offset: '0'
            tooltip: Babystep Z +0.25
            width: '4'
        -   command: M290 Z0.50 P0
            confirmation: false
            enabled_while_printing: true
            icon: fas fa-long-arrow-alt-up
            input: []
            label: Babystep Z +0.50
            message: ''
            offset: 0
            tooltip: Babystep Z +0.5
            width: '4'
        -   command: M290 Z0.75 P0
            confirmation: false
            enabled_while_printing: true
            icon: fas fa-arrow-up
            input: []
            label: Babystep Z +0.75
            message: ''
            offset: '2'
            tooltip: Babystep Z +0.75
            width: '4'
        -   command: M290 Z 1.0 P0
            confirmation: false
[... 828 lines cut ...]
                enabled: true
                faIcon: fa fa-ban
            -   domId: '#tab_plugin_stats_link'
                enabled: true
                faIcon: fa fa-bar-chart
            -   domId: '#tab_plugin_webcamstreamer_link'
                enabled: true
                faIcon: fa fa-video-camera
            -   domId: '#tab_plugin_prettygcode_link'
                enabled: true
                faIcon: fa fa-eye
            -   domId: '#tab_plugin_arc_welder_link'
                enabled: true
                faIcon: fa fa-circle-o-notch
            -   domId: '#tab_plugin_SpoolManager_link'
                enabled: true
                faIcon: fa fa-eercast
            -   domId: '#tab_plugin_consolidate_temp_control_link'
                enabled: true
                faIcon: fa fa-gamepad
            -   domId: '#tab_plugin_eeprom_marlin_link'
                enabled: true
                faIcon: fa fa-list-alt

```

jneilliii, 2020-11-26:
Not sure yet how it will impact buttons with inputs, but I've modified the css to make the button 100%. There's probably a better way to fill the space of the scaffolding box.

```
https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/archive/0.1.7rc2.zip
```

KenLucke, 2020-12-15:
Something broke in the OP 1.5.x update.  6 width buttons no longer are 6x

[image]

(see previous screenshot when things were working:)
https://user-images.githubusercontent.com/57936294/100391411-300cde80-2fe8-11eb-801c-8ae42157d037.png

jneilliii, 2020-12-15:
That's core OctoPrint bootstrap stuff so not sure why that would have been effected unless there's something tht was introduced into the plugin. If you right-click one of those and inspect does the class name have `span6`?

jneilliii, 2020-12-15:
Actually looks like it might be similar to #24.

KenLucke, 2020-12-15:
[image]

KenLucke, 2020-12-15:
BTW, is there a way to tell if a plugin has been abandoned?  The "Active Filters" plugin (that saves the status of the terminal filters) has had no responses from the author for months, even with two requests (by others, from clear back in June) to upgrade it to use Python 3.  It is now completely broken with OP 1.5x and Python 3.

It'd be great if you, @CP2004, or @eyal0 (my  three favorite authors) could take it over, or write an equivalent.

jneilliii, 2020-12-15:
I did one of those a long time ago. It's called Active Filters Extended.

jneilliii, 2020-12-15:
actually the span6 class is probably specified in the parent div, not the button itself, and it appears that is cutoff in your screenshot. I will eventually get around to looking at this again, it's just a matter of spare time.

jneilliii, 2020-12-15:
And relative to your abandonment question, you may want to post an issue, there are the ones we have listed currently. https://github.com/OctoPrint/plugins.octoprint.org/issues?q=is%3Aopen+is%3Aissue+label%3A%22abandoned+plugin%22

KenLucke, 2020-12-15:
> actually the span6 class is probably specified in the parent div, not the button itself, and it appears that is cutoff in your screenshot. I will eventually get around to looking at this again, it's just a matter of spare time.

Sorry

[image]

jneilliii, 2020-12-15:
Yep, so it's assigning the width to the parent, but the css defined [here](https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/blob/6293a7859b72bc8a3b6096d8c9dce90f11c6112e/octoprint_terminalcommandsextended/static/css/terminalcommandsextended.css#L17-L19) appears to not be applying to the button properly. It should be 100%. So not sure if this is a Themeify override that's blocking the default css configuration or not.

KenLucke, 2020-12-15:
I'll disableThemeify and try.  Back in a few.

KenLucke, 2020-12-15:
Nope. Disabled all three items (Theme, Customization, and Tab Icons) in Themeify, no change (other than losing theme colors).  Restarted OP, still no change,  Disabled Themeify in the Plugin Manager and restarted OP, still no joy.

KenLucke, 2020-12-15:
Strange that it's only affecting my 6s...

jneilliii, 2020-12-15:
Ok, I'll experiment with it and give it a go. What's your theme selected?

KenLucke, 2020-12-15:
> Ok, I'll experiment with it and give it a go. What's your theme selected?

Discoranged, but with modificcations.  However, as I said, completely disabling Themeify and restarting Octoprint failed to resolve the issue so I don't think that's it.

jneilliii, 2020-12-15:
You sure you still have the 0.1.7rc2 installed? I Just tested on my pi and it's working with width 6.

[image]

I'm wondering if you add the following to your Themeify if it would fix it?

selector: #control-terminal-custom .btn
css rule: width
value: 100% !important

KenLucke, 2020-12-15:
I have 0.1.6 installed.  Never saw the .1.7rc go by

KenLucke, 2020-12-15:
> I'm wondering if you add the following to your Themeify if it would fix it?
> 
> selector: #control-terminal-custom .btn
> css rule: width
> value: 100% !important

Nope

KenLucke, 2020-12-15:
> I have 0.1.6 installed. Never saw the .1.7rc go by

Found it, but it will be several hours before I can install and test it.

KenLucke, 2020-12-16:
> > I have 0.1.6 installed. Never saw the .1.7rc go by
> 
> Found it, but it will be several hours before I can install and test it.

I just dragged out the spare Pi and did an update on my backup system.  Installing your RC above solved the issue.  I don't know how I missed the RC release, but I apologize for the mistaken bug report.

closed 2022-05-28 by jneilliii