jneilliii/OctoPrint-TerminalCommandsExtended #22: [Feature Request] Colored Buttons

KenLucke, 2020-11-23:
Could you add a parameter for each button to specify the color?    Either via a color picker, or similar to the Macro plugin, which has a dropdown with 5 or 6 selections of color for each button?

As always, thanks for what you do.

jneilliii, 2020-11-24:
Duplicate #11

KenLucke, 2020-11-24:
Not the same request as #11 - that had to do with changing colors when  printing.  I want to be able to set button colors (red, yellow, green, blue, etc.) so that it groups buttons or highlights them.

jneilliii, 2020-11-24:
sorry, forgot the content of that request and was going by issue title.

KenLucke, 2020-11-24:
> sorry, forgot the content of that request and was going by issue title.

No prob, I  have no doubt there's lots on your plate.

jneilliii, 2020-11-24:
You can say that....

[image]

htrabbit, 2022-05-29:
colored buttons would be a great addition