jneilliii/OctoPrint-TerminalCommandsExtended #24: Buttons are no longer compact

EagleB3, 2020-12-01:
**Describe the bug**

Before Octoprint was updated to version 1.5.0, the buttons on the tab were displayed very compactly and it was convenient.
Immediately after the update to 1.5.0, the buttons became huge and do not fit into the previously reserved area of the bookmark. Updating the plugin to versions 0.1.6 and 0.1.7rc2 did not solve this problem.

**Expected behavior**

I really liked the compact layout of the buttons, and I hoped that there would be no changes when changing the version, since the compact presentation of the buttons was very convenient to use.
[image]

_**Maybe it is possible to give the buttons a minimum width (by the size of "Icon + text") when specifying "Width = 0" in the settings?**_

**Screenshots**

I don't have the old screenshot left. All buttons were assigned "**Width = 2**" and on the tab they looked like this:
[image]

Immediately after the update, they turned into the following:
[image]

[image]

After the upgrade, the best I can achieve is:
[image]

**Browser errors**

There are no errors in the browser. There are warnings like "Expected media function name, but found 'touch-enabled'.". Not sure if this is relevant to the problem
[image]

EagleB3, 2020-12-01:
I will also note here that now the field with parameters is always rendered as a column. Even increasing the width does not make it possible to extend the button with parameters in one row. And there is no way to specify that it should be displayed on one line. It seems to me that this is not good:
[image]

If the width is set to "0" and if the value is empty in the "Width" field, the button dimensions are displayed the same, for the entire width of the screen. If you find it possible to implement buttons with a minimum width, then the value "0" can be interpreted as "minimum width", and an empty field = "no width limitation", = "maximum possible width".
[image]

There is no way to reduce or somehow define unreasonably wide gaps between buttons:
[image]

I'm not sure if everything I've described is plugin defects. Most likely, this is the influence of the interface components reworked in version 1.5.0.
It is even possible that users of "tap" displays will be delighted with them. But (my humble opinion) for users of the classic interface with a display and a "mouse" this is a step backward.

jneilliii, 2020-12-01:
I'm not sure @EagleB3 but I like you prefer the other layout for buttons. The only thing I can think of that may have effected this was the changes related to ordering the buttons and adding the new break options. Could also be some upstream change to OctoPrint, which will require some research. I'll see what I can do.

EagleB3, 2020-12-01:
Thank you very much!
If you have any questions, or you need to carry out any checks or testing - please write.

EagleB3, 2020-12-04:
To complete the picture, I rolled back the plugin to ofаicial release 1.5.0 and completely restored the button settings from the my config.yaml backup.
This is how the minimal size buttons that I really like looked like:

[image]

Here is the section describing the "terminalcommandsextended:" from my config.yaml:
[config (TCE-buttons).zip](https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/files/5641871/config.TCE-buttons.zip)

@jneilliii, please tell me, can you download a version of the plugin older than 1.5.0 but earlier than the official release 1.6.0 from somewhere? Now my settings field is completely empty:
[image]
Or is there something else you can do to bring the drag and drop functionality back to "settings"?

jneilliii, 2020-12-04:
Impatient much, most of my plugin work happens on weekends, sorry it's not getting fixed fast enough fast enough for you. If you want to try to fix this, please feel free to fork the plugin, I'm happy to accept pull requests. Unfortunately, the fixes for 1.5.0 octoprint came after my button width changes, so you'd have to incorporate those changes to the 1.5.0 version directly. You can browse through commit history if the rc branch to find the changes.

Ctrl-LeftAlt-Delete, 2023-04-19:
I tried this plugin for the first time today. Thank you [jneilliii] for all the hard work!
I'm on v1.7 and still seeing the same vertical element rendering that [EagleB3] described on Dec 1, 2020.
I manually downloaded/installed version 1.5 and the controls are now rendered horizontally.

jneilliii, 2024-08-03:
I've just released version 0.1.8rc4 that has some tweaks so inputs are inline again instead of stacking. It does require some tweaking to the offset of the button after buttons with input parameters, still trying to figure out if I can fix that.

FajnB, 2024-08-06:
Thank you and sorry for making new issue. I somehow didn't fully realised it was you. Unfortunately with a lot of half way abandoned (not archived) projects the only way to notify the developer seems to create new issue.
So again please accept my apologies, You do an amazing work for the community.

So to the update. It seems to have worked. At least somewhat. It is still very out of proportions.
Still it is the step in the right direction. CSS, js and HTML stuff in general are not my strong sides so I'll try my best to help from a "regular" user experience side of things.

Patronics, 2025-02-04:
> I've just released version 0.1.8rc4 that has some tweaks so inputs are inline again instead of stacking. It does require some tweaking to the offset of the button after buttons with input parameters, still trying to figure out if I can fix that.

0.1.8rc4 is definitely an improvement overall, but, in addition to the offset issue you mention, the button spacing is still really awkward especially when lots of parameters are enabled. However, the spacing of the buttons as they're shown from within the parameter-editing settings pane is much nicer, perhaps it's worth using the layout configuration that are in use there for the rest of the UI as well?

For example, here's the button's appearance from within the settings pane: 

[image]

But here's the same button in the actual interface from the terminal tab (with width 2, or 3, respectively): 

[image]

[image]

Both widths 2 and 3 go beyond the terminal tab's bounding box, yet the settings pane view fits nicely in the same width.

Alternatively, the option to edit the width of each parameter individually would also improve the behavior available overall. 

More minor but also worth noting, the offset and width values currently only accept integer values, but it would be useful to be able to do finer adjustments, such as a width of offset of 2.5.

jneilliii, 2025-02-04:
> width of offset of 2.5

That's not how bootstrap 2 styling works. It's a fixed 12 column layout and parts are expected to be full integers because it's just applying a style name iirc. 

I may be looking at implementing the interface from the new custom control manager plugin bundled in OctoPrint version 1.11.0, which would hopefully resolve all of this.