jneilliii/OctoPrint-TerminalCommandsExtended #25: Ability to set which tab these get attached to (Terminal\Control\Temp\Etc)

dapostol73, 2021-01-08:
Hello, this is more a feature request and I'm new here; if there is a better way to add feature request please let me know.

This tool is great and has help me stream line some workflows, I find my self jumping between octoprint tabs to use it when I'm connect remotely as I want to see my Camera when I run the command.  In the future it would be great if the commands could be associated to a specific tab instead of lock to the Terminal.  For example to be able to extend the buttons on the Control Tab.

As example:

[image]

jneilliii, 2021-01-08:
Yeah, there are other plugins for the control tab, which is why I didn't think it prudent to make it for all. Technically, you don't even need the plugin linked below to do that, it's just a little easier than manually editing the config.yaml file as described [here](https://docs.octoprint.org/en/master/features/custom_controls.html#sec-features-custom-controls).

https://plugins.octoprint.org/plugins/customControl/

jneilliii, 2021-01-08:
Another option would be my Consolidate Temp Control plugin or Consolidated Tabs plugin.

closed 2021-01-08 by jneilliii