jneilliii/OctoPrint-TerminalCommandsExtended #27: Conflict with plugin Display Layer Progress

erickvd, 2021-06-10:
Hello.

I installed both "Terminal Commands Extended" and "Display Layer Progress" on my octoprint.

Display Layer Progress is supposed to show on the web interface the same information as the one on the screen of the printer.

When I activate "Terminal Commands Extended" , the 'Printer Display' only shows the Title.
When "Terminal Commands Extended" is disabled, the info are correctly displayed.

Here are my versions:
Octoprint : 1.5.3
Terminal Commands Extended : 0.1.6 
DisplayLayerProgress : 1.26.0

Regards,

Eric

jneilliii, 2021-06-10:
My plugin should have nothing to do with M117 messages that DLP plugin uses. If there is a conflict it would be for an unknown reason. I'll see about installing both of these on a system and test. When you say `the 'Printer Display' only shows the Title.` what do you mean? The printer says something like Ender 3 ready and never changes during print progress?

erickvd, 2021-06-11:
I did a reboot of my Raspberry Pi after your message.
Problem seems to be solved.

I close this issue.
Sorry for inconvenience.

closed 2021-06-11 by erickvd