jneilliii/OctoPrint-TerminalCommandsExtended #30: The move button seems not to be working.

leoheck, 2021-11-29:
The move button seems not to be working. Am I using it right?

[image]

I am using this version `0.1.6`

How do I move the `Home-XYZ and FL` button to the first place (top-left)?

jneilliii, 2021-11-29:
Yeah, I've been meaning to go back and work on the orientation of these, but you should be able to hold down the move button and then drag it into another area in the list.

leoheck, 2021-11-30:
>  but you should be able to hold down the move button and then drag it into another area in the list.
Well, this is not working. Maybe it is browser-related? I am using Chrome

leoheck, 2021-11-30:
Ah, the same happens on Firefox.

jneilliii, 2021-11-30:
Can you please share your octoprint.log so I can get a better picture of all the things you have installed? There could potentially be a plugin conflict with another plugin.

jneilliii, 2021-11-30:
Actually looking back through release notes, this is a bug with current stable version. If you change the plugin's release channel to `Release Candidate` in OctoPrint's Software Update settings and upgrade when prompted the issue should be resolved.

leoheck, 2021-11-30:
Ah, nice. I will do that.

closed 2021-12-22 by github-actions[bot]