jneilliii/OctoPrint-TerminalCommandsExtended #34: Feature request: Move command buttons above commands filters

Luix333, 2022-06-07:
First I want to thank the developer for this amazing plugin! This is by far my most used plugin. 
I would like to request the option to move the buttons between the terminal textbox and terminal filters. Would this be possible?

[image]

[image]

jneilliii, 2022-06-07:
I think I'd be more apt to move those under advanced. You can test by opening your browser's developer tools and in the console tab enter `$('#terminal-filterpanel').prependTo('#term > div:nth-child(3) > div:nth-child(2)');` to get this.

[image]

[image]

jneilliii, 2022-06-07:
event better...

```
$('#terminal-filterpanel').parent().prependTo('#term > div:nth-child(3) > div:nth-child(2)');
```

Luix333, 2022-06-08:
this looks amazing: 
[image]

jneilliii, 2022-06-09:
Added with the above commit. Go to Software Update section of OctoPrint Software Update settings and change the plugin's release channel to Release Candidate and upgrade when prompted. After enabling the option in the plugin's settings you'll need to refresh your browser.

Luix333, 2022-06-10:
Thanks!!! <3

Luix333, 2024-02-19:
Hi, first of all, thanks for this plug-in and you hard work!
Apparently, something on the latest version of octoprint broke this. Can you please look into this at your earliest convenience?
Thanks again for your work and time!

jneilliii, 2024-02-19:
do you mean in the latest 1.10.0rc2 release?

Luix333, 2024-02-26:
Sorry, I should have specified. Yes, I'm on 1.10.0rc2 
[image]

[image]

jneilliii, 2024-02-26:
updated in above commit and released as version 0.1.8rc2.

Luix333, 2024-04-24:
Hi, thank you again for fixing it last time. I'm sorry to inform that the latest release of octoprint (1.10.0) broke it again. Please take a look at it if you can. Thanks again for your time and effort!

jneilliii, 2024-04-24:
updated in the latest rc version.

closed 2024-04-28 by Luix333