jneilliii/OctoPrint-TerminalCommandsExtended #35: Command parameters are rendered incorrectly

FajnB, 2024-08-03:
This is a direct response to issue #24 as it is unfortunately wrongly marked as enhancement when it should be marked as bug and also went completely stale.

I do respect your efforts and that this is a hobby and not a full time job. I am thankful for your work.
What I was trying to say is that the above mentioned issue is a bug as even in your documentation about the plugin it shows command parameters clearly positioned next to the command button. Viz.: [link](https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended/blob/master/screenshot_tab.png)
Right now the command parameters are displayed above the command button and also the input box is in most cases out of proportions which stretches the command button as well.
Also in the "Terminal Command Editor" it is previewed correctly.

This is how it is supposed to look like (from "Terminal Command Editor"):
[image]

This is how it looks like in the Terminal tab:
[image]

jneilliii, 2024-08-03:
> This is a direct response to issue #24 as it is unfortunately wrongly marked as enhancement when it should be marked as bug and also went completely stale.

Them you should comment on that issue and not create duplicates.

closed 2024-08-03 by jneilliii

jneilliii, 2024-08-03:
I've just released version 0.1.8rc4 that has some tweaks so inputs are inline again instead of stacking. It does require some tweaking to the offset of the button after buttons with input parameters, still trying to figure out if I can fix that.